ghent360/PrntrBoardV2 #1: Schematic library missing

cruse123, 2019-11-07:
Symbols in schematics cannot be normally displayed, showing rectangles with question marks. When double-clicking the part and the icon button at "Symbol - Library Reference", an error message prompts saying `Library file '.....\PrntrBoardV2\hardware\10018784-10210TLF.lib' not found. Use the Manage Symbol Libraries dialog to fix the path (or remove the library).` Maybe there should be a library file named "10018784-10210TLF.lib" but not uploaded to the repository?

Screenshots are as follows:

[image]

[image]

Many thanks!

ghent360, 2019-11-07:
Thank you for reporting the issue with the schematics. I'll take a look and try to improve the content of the repo, so it is self contained.

As far as "10018784-10210TLF.lib" - it is not needed and while there is a warning on my system as well. I'm able to see the schematics, so that is not what is causing the problem for you.

It is very inconsiderate of me to leave a broken library reference in the sym-lib-table. I'll remove it asap.

ghent360, 2019-11-08:
Hello, would you kindly let me know, what version of KiCad are you using?
I tried to download the project on a different computer and I was able to view the schematics with no issues with version 5.1.4 (latest stable build). Unfortunately that version was unable to display the PCB, for that you need the nightly development branch.

I did verify that after installing the build from today (11/7/2019) I was able to view both the schematics as well as the PCB.

For convenience I added a PDF of the schematics in the hardware/doc folder.

cruse123, 2019-11-08:
Hello!
After pulling the new commits, schematic symbols can be normally displayed, without any errors at all! I was using the latest stable release `kicad-5.1.4_1` at that time and everything in schematics went well. Problems solved. Thanks a lot!

However, as you have mentioned, PCB layout document cannot be opened in this "lower" version. Switching to today's latest nightly build `kicad-r14425.6182133a4` works. If a stable version is preferred, manually modifying the `kicad_pcb` document may help: Expand error information in the kicad dialog, delete the "Unexpected" lines in the PCB document with text editors. This may be an imperfect solution, so for read-only previewing only.  Using the proper version for further editing / manufacturing is recommended in my opinion.

Thanks again!

closed 2019-11-09 by ghent360