ghent360/PrntrBoardV2 #3: RRF compatibility

ArkadiuszRaj, 2021-01-14:
Hello, 

First of all congrats for this work.
As a foreword I want to express my openness in helping porting of the STM RRF to support this board. I am interested in assembling of this board for my purposes but to do so (make RRF running) I think small modifications are necessary. 

- enable CAN to allow using Duet3 modules. This can be done by small remapping of GPIO (PD0-1 for CAN, move ST5/6_STEP to PE7-8 and combine U_MIN/Z_MAX inputs with SWDIO/SWCLK lines)
- resign from EXP1/EXP2 in favor of 4-pin TFT screen connection (like PanelDue) as I think those EXPs are ancient interface nowadays:)
- add Wi-Fi by mimicking available adapters of NodeMCU/esp8266

I plan to use your design in future to control my Voron2.4 printer, which uses 6 motors to drive 3 axes. Thus I will need CAN to talk to Duet3 1LC for hotend. That is simpler than supporting 8x PCIe slots. 

Why DC converter module is not vertically installed using something like 2.54 2x4 headers? Can you share background of this decision to place it horizontally?

I also think adding second converter in parallel, converting VIN to 12V to allow selection of this voltage to drive fans - much much cheaper fans as broadly available on computer market. 

BTW, have you verified PCB with STM32F7 cpu? 

Appreciate your comments. 

Regards 

Arkadiusz

ghent360, 2021-01-14:
Hi Arkadiusz,

I'll think about adding CAN interface, thank you for the suggestions. There is a spare SPI and serial and I2C interfaces on EXP3 port. Between these and the EXP1/EXP2 GPIOs it should be possible to connect serial TFT display and Wi-Fi module.

The boost converter module I use has castellated edges and solders onto the board very easily (https://www.aliexpress.com/item/32713451014.html?spm=a2g0s.9042311.0.0.27424c4dCd0jRT). I found that the cost of this module is less than the cost of parts I would need to add one to the board. You can add external 12V regulator and feed the 12V info the middle of the fan voltage selector pin. This would make the fans work at 12V. The MOSFETS are rated for up to 40V, so I would not go higher than 30V on that pin.

I have not considered the F7 chips, because they are much more expensive and I don't think the existing software utilizes the F4 to the level that would justify the cost increase. I did make a version with an H750, chip (new_cpu branch). That however needs a more complex bootloader as the chip does not have enough flash to store the firmware and it needs to be loaded from SPI flash or sd-card into RAM.

I do use the JLCPCB assembly service to order partially populated board. The KiCAD file has all component references for the JPC part numbers that are needed. The boards come without the micro-sd card connector, no micro usb connector and no DC buck converter. You would also need to solder all true hole connectors you care about. It takes me about an hour to assemble one board. The JLC service has a limit on the variation of components one can place on the board, which is one more reason to skip on the DC converter and use premade one.

Happy to chat more.

ArkadiuszRaj, 2021-01-14:
Hello

Yes I saw this specific branch regarding H7 series chip. My mistake with
wrong character:)

I have already updated schematic with CAN interface, not yet on pcb though.
Will share image later on.

Also while using JLCPCB I would swap 0603 into 0402 to save some space.
However keeping 2 layer design from one hand make manufacturing them
cheaper but is much more challenging from routing perspective. Must admit
that for some time I am using JLCPCB for 4-layer PCBA but with PCB not
exceeding 10x10cm.

So changing to 0402 and some cleaning up of tracks. That is the way forward
to my understanding.
Also when it will be possible to cut some mm from both dimensions this
design can jump into 4 layers and compact size even more.

Wi-Fi is the least problematic addon.

ArkadiuszRaj, 2021-01-14:
As promised, CAN interface  as simple as possible:
[image]
and gpio remapping...
[image]

Actually diodes connecting U_MIN/Z_MAX with SW lines are to me not necessary as I presume SWD port is used only for initial flashing or possibly never at all.

BTW, with 0402 there is possibility to fit some resistors/diodes that I saw outside the board (full_prot branch):

[image]
this is excerpt of the PCB close to J12...

and  BAT54S...
[image]

ghent360, 2021-01-14:
Most of the board size is dictated by the external connectors. 10x10cm would be very challenging. My V1 design was using that, but the layout of the connectors was very impractical.

ghent360, 2021-01-15:
FYI the extClock is the most up-to-date branch.

I have another idea for scrounging a few extra pins: the sdio can work in 1-bit mode, so we can re-task the SDIO_D2, D3 and D4 pins. As a practical matter it is not stable in 4-bit mode with the current layout anyway - I suspect the signal return path is quite bad with only 2 layers.

ArkadiuszRaj, 2021-01-15:
Hi

Thanks for pointing proper branch for changes. Actually this is something I wanted to ask you. 

Referring the connector placement and board ergonomics I agree. I even think one can scarify additional few euros to have 4 layer board (8 vs 32). This way the pcb can have better power distribution and shielding characteristics. Yet 4-layers can be treated as roadmap item. Now the important part is CAN to me. 

I will then update schematic on the pointed branch and create PR. 

For the board itself as for now I will only do cleanup of the tracks what shall make life easier when change to 0402 happen.

ghent360, 2021-01-15:
Do you want to place the can i/o on the board or as add-on via header?

ghent360, 2021-01-15:
Btw I want to remove the connector next to the micro-sd card. It does not
serve any useful purpose.

ArkadiuszRaj, 2021-01-15:
Hi,

1) WiFi using add-on board and utilizing EXP1/EXP2, simimalar to SKR RRF WiFi, but in a shape of [SKR GTR adapter ](https://www.tindie.com/products/pcr/reprapfirmware-wifi-adapterboard-for-gtr/)

2) CAN - albeit I prefer integration it can also be an add-on. But in this case a PCB must be modified to release CPU pins dedicated to CAN interface.

3) I did a clean up of the tracks and added 5V terminal to use for installations, where external power supply is controled by board, i.e. cut off of the VIN is controlled by gcode. BTW I think I saw somewhere in branches PSU output, or am I mistaken?
Anyway this is how it looks, I had to reorder overvoltage protection components.
[image]

4) You want to remove EXT4? Which as I understand exposes the SDIO lines?
Yes, current routing is not good for high speed communication. This means it realy can be tailored down do simple SPI card access. Frankly speaking, in such a control boards there is no need for high speed communication.

ghent360, 2021-01-15:
Any thoughts on using TJA1040? Allegedly you can skip the common mode choke.

ArkadiuszRaj, 2021-01-16:
Please find corrections in PR https://github.com/ghent360/PrntrBoardV2/pull/4

ghent360, 2021-01-16:
Merged.

closed 2021-01-16 by ghent360