ghent360/PrntrBoardV2 #6: WiFi Addon

ArkadiuszRaj, 2021-01-16:
PR already provided #5 

ESP-12F used, to allow usage of DWC. Connected to main board via SPI2 exposed on EXT1/2 ports
Local USB to easly update firmware of the ESP.
As EXT1/2 do not provide 3V3 I have put also local dc-dcconverter.

[image]

[image]

Please observe PR has by mistake main project files (schematics & pcb) that I have used as template. Please reject changes on those files. I.e. this PR shall have only new project files plus ESP component libs (part, footprint & 3d view)

ArkadiuszRaj, 2021-01-16:
If stacking for legacy LCD would be needed I sugegst to use long led 5x2 header sockets

ArkadiuszRaj, 2021-01-17:
Correction to Header placement and spacing applied. Sorry for missing checking such a basic stuff

ArkadiuszRaj, 2021-01-17:
Regarding RST - it is essential and shall be connected. 

My bad I have not checked the EXP ports.

ghent360, 2021-01-17:
Well, not your fault. I just noticed it by chance when trying to see if the
connector of the addon lines up with the connector on the board. Again
thank you for the effort.

ghent360, 2021-01-17:
I think this PR is done and we can close it. We can fix the EXP2.4 pin in the main board schematics as separate issue.

closed 2021-01-17 by ghent360