ghent360/PrntrBoardV2 #10: sd-card i/o rework & fix

ghent360, 2021-01-17:
At the moment the sd-card connector is broken (missing CS pin connection).
We can:

1. Move the sd-card to use hardware SPI1, freeing the SPI3 pins.
2. Scrounge some new pin for the CS connection
3. Switch the sd-card back to 1 bit SDIO and scrounge for pins.

Downside of #1 and #2 is that I need new bootloader to read sd-card via SPI. Not a terribly big issue.
Downside for #2 and #3 is that we are short on pins again &^%@#$%^!

Best to probably byte the bullet on the bootloader and do #1, we'll have a few extra IO pins, which is nice.

ArkadiuszRaj, 2021-01-18:
Vote for option 1, i.e move sd card tp spi1.

Also Iam interested in paricipating in IAP bootloader development. 
But to do so, I need first a board itself. However I want to await this as well as #9 issue is solved.

ghent360, 2021-01-18:
Code for the bootloader:
https://github.com/ghent360/STM32F4_Bootloader

ghent360, 2021-01-18:
Re-wired the SD-card to SPI1 in the schematics and added extra pins to EXP2 and EXP3. Removed 3.3V from EXP3.

Need to route the changes in the PCB.

ghent360, 2021-01-18:
It has been routed. Thank you @ArkadiuszRaj

closed 2021-01-18 by ghent360