ghent360/PrntrBoardV2 #12: StepStick driver considerations

ArkadiuszRaj, 2021-01-17:
Did another small thing.

[image]

Questions:
- shall DIAG output of the stepstick (TMC one) be connected to Alarm1?
- what is purpose of CFG1-3 pins on the PCI slot? i.e PrntrBoard do not use them.

ArkadiuszRaj, 2021-01-18:
I saw your schematic changes and as I have some spare time I have played a bit with routing.

[image]

[image]

4 layers, one used to distribute V_IO.

That is preliminary view. I have added more vias around cap ground.

ghent360, 2021-01-18:
I added one more jumper and routed it in 2 layers. May need some more vias
for the GND. Add your handle to the Copyright, it is mostly your design.

ArkadiuszRaj, 2021-01-18:
PR #13 ready

ghent360, 2021-01-18:
Merged.

closed 2021-01-18 by ghent360