ghent360/PrntrBoardV2 #8: Folders structure enhancement

ArkadiuszRaj, 2021-01-17:
I suggest to add following folder structure
- (one folder per product)

  - Manufacture

    - schematics.pdf
    - ibom.html
    - gerbers.zip
    - bom.csv
    - placement.csv

This is complete data for the ordering boards as well as assembly. Such a product manufacture' subfolder then represents "release" data content. If you agree I can start with making order following this suggestion.

ghent360, 2021-01-17:
The "Manufacture" folder sounds like a great idea. Since there are multiple
boards in this depot, we have to decide how to proceed:
Option 1) have board prefix: prntrboard_v2_schematics.pdf,
prntrboard_v2_ibom.htmletc for the main board; then tmc2209_schematics.pdf
tmc2209_ibom.html etc.
Option 2) have subfolders: Manufacture\prntrboardv2; Manufacture\tmc2209
etc.

I personally favor #2 since with #1 the list of files would be quite large
and hard to manage what is what, but I'm open to other suggestions.

~V

ArkadiuszRaj, 2021-01-17:
Option #2 follows my idea.

I will refactor files but will not add jlcpcb assembly files. Have no idea
how you create those and frankly speaking it is painful process as they do
not follow any standard of initial orientation. Thus every project one must
check if components orientation is OK.

ghent360, 2021-01-17:
I have a magic awk script that fixes the orientation on components I know
are wrong. JLC people are nice and will fix wrong orientation for you even
if you get it wrong, just have some cues in the silkscreen about the
correct orientation.

ghent360, 2021-01-17:
Yes I agree one has to check the orientation. Ill add the manufacture files
I have used and know to work well.

ArkadiuszRaj, 2021-01-17:
PR pushed, now the ball is yours 👍 

What I am thinking about is that new version of for example TMC2209 driver shall not appear as additional file set but the repository shall by flagged with tags. Then information about revision in README file shall be updated keeping old data untouched.

ghent360, 2021-01-18:
I changed the StepStick driver schematics a bit. Removed the external voltage source - I'm not aware of any drivers that can be driven with over 24V successfully. Added local filter cap and a few jumpers to account for a variety of BTT and other drivers.

ghent360, 2021-01-19:
I updated all the driver boards.

closed 2021-01-19 by ghent360