ghent360/PrntrBoardV2 #9: Fix EXP2 pin 4

ghent360, 2021-01-17:
EXP2 pin 4 is currently not connected. With the recent addition of the CAN interface we have spare GPIO pins and we can add a connection for that pin.

This pin is needed for the WiFi add-on board.

ArkadiuszRaj, 2021-01-18:
You have written:
"In the "reprap" screens that pin is usually used to detect if SD-card is inserted in the LCD sd-card slot. This is optional feature that I skipped because I was running out of pins."

~SD_DETECT is now connected to EXP2.7 which is connected to SD card switch and then to PD3 port.
So it is not a signal from the LCD screen own SD card. If one will use such a screen you think mixing of two SD_DETECTs is OK?

RRF Config tool for SKR1.4 produces
```
//ESP Settings
8266wifi.espDataReadyPin = 0.28;
8266wifi.lpcTfrReadyPin = 1.30;
8266wifi.espResetPin = 1.31;
//ESP RX/TX Settings
8266wifi.serialRxTxPins = { 4.29, 4.28 } ;
```
for GTR
```
//ESP Settings
8266wifi.espDataReadyPin = A.15;
8266wifi.lpcTfrReadyPin = C.11;
8266wifi.espResetPin = B.10;
//ESP RX/TX Settings
8266wifi.serialRxTxPins = { PC7, PC6 } ;
```
"ESP RX/TX Settings " is used solely for ESP firmware upgrade.

Anyway, for reset pin.

For SKR 1.4 - 1.31 is GPIO line connected to EXP2.4
For SKR GTR - B.10 same CPU as here, connected as well to EXP2.4, on schematics called EXP_SW_B10.

EXP SD_DETECT handling.

For SKR 1.4, EXP2.7 is connected directly to SSEL0 port of LPC
For SKR GTR, EXP2.7 it is EXP_SS_PB12 signal

TF card SD_DETECT:

For SKR 1.4 it is P0.27 line
For SKR GTR is is TF_SW connected to PC4

So as you can see, EXP SD_DETECT is different than "TF_DETECT".

Then I suggest:
- Left PD3 for internal SD_DETECT
- Assign PD2 for EXP_SW_RST (pin EXP2.4)
- Reuse two SWD pins as GPIO, this requires high imp buffer to not mix SWD with GPIO externally (or to make SWD functional) 
- Use one of the SWD pins for connecting EXP2.7 (EXP_SD_DETECT)
- And still have one SWD/GPIO pin as buffer

ghent360, 2021-01-18:
Will be taken care of with issue #10 .

closed 2021-01-18 by ghent360