ghent360/PrntrBoardV2 #14: StepStick better air flow

ghent360, 2021-01-18:
The distance between driver boards mounted vertically is 20mm. With the StepStick adapter, this does not leave a lot of room for heat sink on top of the driver. In addition on the bottom there is a wall of jumpers used for SPI/CFG switching, which would prevent air flow and installing heat sink on the bottom.

We should consider moving the SPI/CFG jumpers out of the way. Not a trivial task because there is not a whole lot of real estate left on the board. This will also break how these are traditionally organize on RAMPS boards and will require extra documentation.

ArkadiuszRaj, 2021-01-18:
Well, I think there can be space for moving out this pin forest.

[image]

But... I have doubts, as this will only fix for those , that have IC mounted on the bottom side. But majority of the older polulu drivers have chips on top side. Here, to make space for regular heatsink, maybe one can you low profile pin sockets? like these one https://www.aliexpress.com/item/622209657.html

ghent360, 2021-01-19:
Done.

closed 2021-01-19 by ghent360